toblum/ESPTeamsPresence #2: Trouble compiling in PlatformIO

enochrod, 2020-08-16:
I was trying to compile the code in PlatformIO, but I ran into some errors.  Please see below.  Can you please provide some assistance?  

- I only want the light to come on when I am busy, so I want to make this change.

- I also need to adjust the color of the light when I am busy.  For some reason it is showing up as purple, stead of red.

[image]

**Errors**
```
In file included from C:/users/rk/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/freertos/freertos/FreeRTOSConfig.h:120:0,
                 from C:/users/rk/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/freertos/freertos/FreeRTOS.h:99,
                 from C:/users/rk/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:32,
                 from src\main.cpp:13:
C:/users/rk/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp32/rom/ets_sys.h:21:21: fatal error: soc/soc.h: No such file or directory
compilation terminated.
In file included from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos/freertos/FreeRTOSConfig.h:120:0,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos/freertos/FreeRTOS.h:99,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_event/esp_event.h:20,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32/esp_event_loop.h:22,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiGeneric.h:27,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiSTA.h:28,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFi.h:32,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\ETH.h:24,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\ETH.cpp:21:
C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32/rom/ets_sys.h:21:21: fatal error: soc/soc.h: No such file or directory
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib641\WiFi\ETH.cpp.o] Error 1
In file included from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos/freertos/FreeRTOSConfig.h:120:0,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos/freertos/FreeRTOS.h:99,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_event/esp_event.h:20,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32/esp_event_loop.h:22,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiGeneric.h:27,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiSTA.h:28,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFi.h:32,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFi.cpp:24:
C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32/rom/ets_sys.h:21:21: fatal error: soc/soc.h: No such file or directory
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib641\WiFi\WiFi.cpp.o] Error 1
In file included from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos/freertos/FreeRTOSConfig.h:120:0,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos/freertos/FreeRTOS.h:99,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_event/esp_event.h:20,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32/esp_event_loop.h:22,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiGeneric.h:27,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiSTA.h:28,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFi.h:32,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiAP.cpp:25:
C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32/rom/ets_sys.h:21:21: fatal error: soc/soc.h: No such file or directory
compilation terminated.
In file included from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos/freertos/FreeRTOSConfig.h:120:0,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos/freertos/FreeRTOS.h:99,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:32,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiClient.h:24,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiClient.cpp:20:
C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32/rom/ets_sys.h:21:21: fatal error: soc/soc.h: No such file or directory
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib641\WiFi\WiFiAP.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib641\WiFi\WiFiClient.cpp.o] Error 1
In file included from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos/freertos/FreeRTOSConfig.h:120:0,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos/freertos/FreeRTOS.h:99,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_event/esp_event.h:20,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32/esp_event_loop.h:22,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiGeneric.h:27,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiSTA.h:28,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFi.h:32,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiSTA.cpp:25:
C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32/rom/ets_sys.h:21:21: fatal error: soc/soc.h: No such file or directory
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib641\WiFi\WiFiSTA.cpp.o] Error 1
[... 45 lines cut ...]
*** [.pio\build\esp32doit-devkit-v1\lib641\WiFi\WiFiGeneric.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib641\WiFi\WiFiScan.cpp.o] Error 1
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib641\WiFi\WiFiUdp.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib641\WiFi\WiFiServer.cpp.o] Error 1
In file included from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos/freertos/FreeRTOSConfig.h:120:0,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos/freertos/FreeRTOS.h:99,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:32,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src/WiFiUdp.h:38,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\DNSServer\src\DNSServer.h:3,
                 from C:\users\rk\.platformio\packages\framework-arduinoespressif32\libraries\DNSServer\src\DNSServer.cpp:1:
C:\users\rk\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32/rom/ets_sys.h:21:21: fatal error: soc/soc.h: No such file or directory
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\libd45\DNSServer\DNSServer.cpp.o] Error 1
[... cut ...] =

Environment            Status    Duration
---------------------  --------  ------------
az-delivery-devkit-v4  IGNORED
esp32doit-devkit-v1    FAILED    00:00:03.601
========================================================== 1 failed, 0 succeeded in 00:00:03.601 ==========================================================
The terminal process "C:\Users\rk\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

toblum, 2020-08-16:
Hi @enochrod,

let me see if I can help:

- It should be very easy for you to change the lighting schema by patching the function setPresenceAnimation(). Just set the color to black like for offline if you don't want the LEDs to light up. https://github.com/toblum/ESPTeamsPresence/blob/7a51cba094689318baff470aaf94f38a364bebfd/src/main.cpp#L277
- Busy is set to purple by default. Just set it as you wish:
https://github.com/toblum/ESPTeamsPresence/blob/7a51cba094689318baff470aaf94f38a364bebfd/src/main.cpp#L290

Unsure about the compilation errors. What version of the ESP32 platform do you have installed in platform.io? I use 1.12.4.

Cheers
Tobias

enochrod, 2020-08-18:
Hi @toblum, I was able to resolve the errors by uploading the missing files to the `\.platformio\packages\framework-arduinoespressif32` directory.

Thank you for pointing me in the right direction for changing the lighting schema.  I made the following changes.

```
void setPresenceAnimation() {
	// Activity: Available, Away, BeRightBack, Busy, DoNotDisturb, InACall, InAConferenceCall, Inactive, InAMeeting, Offline, OffWork, OutOfOffice, PresenceUnknown, Presenting, UrgentInterruptionsOnly

	if (activity.equals("Available")) {
		setAnimation(0, FX_MODE_STATIC, BLACK);
	}
	if (activity.equals("Away")) {
		setAnimation(0, FX_MODE_STATIC, BLACK);
	}
	if (activity.equals("BeRightBack")) {
		setAnimation(0, FX_MODE_STATIC, BLACK);
	}
	if (activity.equals("Busy")) {
		setAnimation(0, FX_MODE_BREATH, RED, 1000);
	}
	if (activity.equals("DoNotDisturb") || activity.equals("UrgentInterruptionsOnly")) {
		setAnimation(0, FX_MODE_SCAN, RED);
	}
	if (activity.equals("InACall")) {
		setAnimation(0, FX_MODE_BREATH, RED, 9000);
	}
	if (activity.equals("InAConferenceCall")) {
		setAnimation(0, FX_MODE_BREATH, RED, 9000);
	}
	if (activity.equals("Inactive")) {
		setAnimation(0, FX_MODE_BREATH, WHITE);
	}
	if (activity.equals("InAMeeting")) {
		setAnimation(0, FX_MODE_SCAN, RED);
	}	
	if (activity.equals("Offline") || activity.equals("OffWork") || activity.equals("OutOfOffice") || activity.equals("PresenceUnknown")) {
		setAnimation(0, FX_MODE_STATIC, BLACK);
	}
	if (activity.equals("Presenting")) {
		setAnimation(0, FX_MODE_COLOR_WIPE, RED);
	}
}
```

closed 2020-08-18 by enochrod

toblum, 2020-08-18:
Good to hear! Have fun with it!