toblum/ESPTeamsPresence #3: Start device login does not generate code - deserializeJson() failed: NoMemory

raddi, 2020-09-09:
Hey,
I just flashed my ESP with your code with platform.io, seems to run fine. After I connect to the AP and fill in all the informations, the ESP does connect to Wifi and is blinking green. When I go to the device setup page and click the "Start device login" button, no code is generated and the device still blinks green. The console output is:

handleStartDevicelogin()
[HTTPS] Method: POST, Response code: 400
deserializeJson() failed: NoMemory

The device page stats under memory usage: Sketch: 233216 of 1319729 bytes free RAM: 235616 of 327680 bytes free - seems okay to me.

Do you have any idea how to solve this?
Thanks in advance :)

toblum, 2020-09-09:
Hi @raddi,

that's no easy to diagnose from here, but es the call in handleStartDevicelogin() seems to return a HTTP 400 instead of a 200, there could be something wrong with this call. This call uses two of your supplied parameters: "tenant" and "client_id". Could you please double-check if these are correct, e.g. no spaces before and after and the correct values are entered?

Did you use the client_id I supplied, or did you create your own app?

If this not helps, you may add some more debug output here: https://github.com/toblum/ESPTeamsPresence/blob/7a51cba094689318baff470aaf94f38a364bebfd/src/request_handler.h#L17 just to make sure, the values are passed correctly:
DBG_PRINT("url:");
DBG_PRINTLN(url);
DBG_PRINT("payload:");
DBG_PRINTLN(payload);

Let me know if that helped.

Best regards
Tobias

raddi, 2020-09-10:
Thank you for your help, I just had the wrong Tenant ID - sorry for bothering you :) It works fine now!

closed 2020-09-10 by raddi

toblum, 2020-09-10:
No problem! Good to hear that it works now!