toblum/ESPTeamsPresence #4: security concern

programmer131, 2020-10-22:
Hi!
Thanks for sharing this project,
it isn't issue, but a concern, will this code store my data/MS account access anywhere other than this esp8266 module? or is it all local?

toblum, 2020-10-22:
Hi @programmer131,

this code wont see and save your credentials (username/password) on the module. The device auth flow is designed like that. The only thing that is saved on the module is an authentication token that allows the code to read your presence info and minimal needed user data. This token doesn't allow anything else.
To be clear: This token can be extracted from the module by experienced people if they have access to the module. The ESP32 has no really save place to store this in a protected manner (this is something for higher priced solutions). But as this token is not too sensitive, I can personally live with that.

If you want to register your own Azure AD app, you do it like it's described here: https://github.com/AzureAD/microsoft-authentication-library-for-dotnet/wiki/Device-Code-Flow#application-registration

I hope this answers your questions. Feel free to write, if you have any questions.

Best regards
Tobias

programmer131, 2020-10-23:
thanks @toblum 
I registered my own app, it works great.

closed 2020-10-23 by programmer131