toblum/ESPTeamsPresence #5: Error connecting to Microsoft account

rasppliances, 2020-11-03:
Hello,

After struggling to get my ESP32 device up and running (the firmware loaded but the device was giving errors, and I don't use VSCode for Arduino, so I had to build my own sketch), I finally got things going.  I got to the Start Device Login point, and went through the steps indicated.  I ended up getting an error from Microsoft:

> AADSTS50020: User account 'P5500221@ness.com' from identity provider 'https://sts.windows.net/65bc0b3b-7ca2-488c-ba9c-b1bebdd49af6/' does not exist in tenant 'Microsoft Services' and cannot access the application '3837bbf0-30fb-47ad-bce8-f460ba9880c3'(Teams presence device) in that tenant. The account needs to be added as an external user in the tenant first. Sign out and sign in again with a different Azure Active Directory user account.

I suspect it has to do with the Tenant hostname/ID I put in.

I'm attaching the startup output, as pasting it in here makes it all weird for some reason.

Thanks
Rick
[startupoutput.txt](https://github.com/toblum/ESPTeamsPresence/files/5479017/startupoutput.txt)

toblum, 2020-11-03:
Hi @rasppliances,

good to hear that you were able to get it working on Arduino IDE too. 

I suspect that the tenant id just isn't correct. What kind of account are you using for teams? Is this a regular O365 business account or something else like teams free?
The tenant name should usually be something like "something.onmicrosoft.com" or a GUID.

Here is some guidance how to determine the tenant ID: https://teams.handsontek.net/2019/04/09/how-to-get-microsoft-teams-tenant-id/

Best regards
Tobias

rasppliances, 2020-11-04:
Thanks for the feedback. You were right, it was the tenant id, and the link you gave was a big help. (PS: It's a business account.)

The problem I have now is that when I try to reconfigure it (to change the number of LEDs), it's asking for a username and password.  I've tried ESPTeamsPresence with the password I setup in AP Password, but that doesn't work.  And I tried my Microsoft account and password, but that didn't work either.

Thanks again.
[image]

rasppliances, 2020-11-04:
I got it!  The username is admin and the password is the ap password set during initial setup.

closed 2020-11-04 by rasppliances

toblum, 2020-11-04:
Hi @rasppliances,

Sorry couldn’t reply earlier. Great you found it out by yourself. I should write this down buffet somewhere...

Greetings