toblum/ESPTeamsPresence #6: Not an issue, but a suggested feature with code

rasppliances, 2020-11-06:
Hi,

Thanks for your help on the last issue.  I've added some code to v1.4 to include a brightness setting.  I thought I'd share it with you.  I've attached my versions of main.cpp, and request_handler.h, and here are the changes I made using the line numbers in my files.  Feel free to add this to your project, or toss it in the bin. It's just something I added, and thought I'd give back.

Rick
[main.cpp.txt](https://github.com/toblum/ESPTeamsPresence/files/5498484/main.cpp.txt)
[request_handler.h.txt](https://github.com/toblum/ESPTeamsPresence/files/5498485/request_handler.h.txt)

- main.cpp
- 	134 char paramBrightnessValue[INTEGER_LEN];
- 	140 IotWebConfParameter paramBrightness = IotWebConfParameter("Brightness (default: 10)", "brightness", paramBrightnessValue, INTEGER_LEN, "number", "1..100", "16", "min='1' max='100' step='1'");  
- 	149 int brightness;
- 	612   iotWebConf.addParameter(&paramBrightness);
- 	628  brightness = atoi(paramBrightnessValue);
- 	629  if (brightness == 0) {
- 	630    DBG_PRINTLN(F("Brightness=0, using 10."));
-	631     brightness = 10;
-	632   }
-	633   ws2812fx.setBrightness(brightness);
- 
- request_handler.h
- 	169 s += "<div class=\"nes-field mt-s\"><label for=\"name_field\">Brightness</label><input type=\"text\" id=\"name_field\" class=\"nes-input\" disabled value=\"" + String(paramBrightnessValue) +  "\"></div>";
-   	207  responseDoc["brightness"].set(paramBrightnessValue);
-  	270  int l5 = server.arg(paramBrightness.getId()).length();
-	271   if (l5 < 1)
-	272   {
-	273     paramBrightness.errorMessage = "Please provide a value for the brightness!";
-	274     valid=false;
-	275   } 
- 	283   ws2812fx.setBrightness(atoi(paramBrightnessValue));

toblum, 2020-11-08:
Thank you @rasppliances. I'll keep that open and have a look at it when I work again at it.

toblum, 2021-01-17:
See: https://github.com/toblum/ESPTeamsPresence/issues/10#issuecomment-761893201