toblum/ESPTeamsPresence #7: Asking for username / password when changing configuration?

SethColeman, 2020-11-13:
When trying to modify the configuration via browser (connected to my home wifi), I get a username / password login popup box.

Serial monitor shows: 
[image]

toblum, 2020-11-13:
Hi @SethColeman,

I need to write that down better. To update configuration you need to login with user "admin" and the password you provided during setup. (Default is "presence")

Hope that works

Greetings

SethColeman, 2020-11-13:
Toblum, Ok, I'll try it out next time I have the issue. I was able to reset the device to defaults and get it to work. When I was having the issue I had an incorrect tenant ID and I was unable to get back in to fix it. I'm just waiting on my IT department to (hopefully) approve my azure request...

toblum, 2020-11-13:
OK, great, feel free to ask, if you have any issues!

closed 2020-11-13 by toblum