toblum/ESPTeamsPresence #10: Cant get status

bjwelker, 2021-01-15:
It work for about 3 days and now I get this.

Polling presence failed, retry #0.
Polling presence info ...
[HTTPS] Auth token valid for 3537 s.
[HTTPS] Method: GET, Response code: 200
deserializeJson() failed: NoMemory
--> Availability: , Activity:

pduck27, 2021-01-15:
Me too, seems that the json is bigger now. Change this line from 220 to 500 works for me.

```
void pollPresence() {

	// See: https://github.com/microsoftgraph/microsoft-graph-docs/blob/ananya/api-reference/beta/resources/presence.md

	// const size_t capacity = JSON_OBJECT_SIZE(4) + 220;

	const size_t capacity = JSON_OBJECT_SIZE(4) + 500; // pduck27
````

toblum, 2021-01-16:
Hi @bjwelker and @pduck27,

thank you for pointing me to this. The beta endpoint was extended. It now contains the out-of-office info. That's why the allocated memory wasn't enough.

```json
{
    "@odata.context": "https://graph.microsoft.com/beta/$metadata#users('c4ccd1af-5e15-5955-86dd-59fd771c81b7')/presence/$entity",
    "id": "c4ccd1af-5e15-5955-86dd-59fd771c81b7",
    "availability": "Away",
    "activity": "Away",
    "outOfOfficeSettings": {
        "message": null,
        "isOutOfOffice": false
    }
}
```

I released the new version 0.16.0 that allocates more memory and also uses the v1.0 endpoint that is now available and should be more stable.

Greetings
Tobias

closed 2021-01-16 by toblum

pduck27, 2021-01-17:
Thank you for fast solution and adding the new endpoint. I updated my fork recently. 

Would you be interested to get a push request for adding brightness control to your original? You can check my fork along to the // pduck27 comments.

Best regards
pduck

toblum, 2021-01-17:
Hi @pduck27,

I already started to adopt your code locally, but I'm unsure if I should add it to the main branch as the config schema needs to to changes for the additional field.
If I just add it, there will be random data in that field that could cause problems for existing users. If I update the config version of IotWebConf, existing users would need to reconfigure everything. Both options are not ideal. So I just added it to a additional branch https://github.com/toblum/ESPTeamsPresence/tree/feature/brightness_control for now. Not sure how to handle this situation good for existing users.

Greetings
Tobias

pduck27, 2021-01-18:
Hi @toblum ,
you are right, I had this issue too when extending my fork. Some kind of upgrade routine would be nice. I will research too if this can be handled somehow.

Thank you
Alex