toblum/ESPTeamsPresence #11: I deleted the APP in MS

Eisbaeeer, 2021-01-16:
Hi,
how can I get the APP registering twice?
I deleted/not allowing the App in O365 portal and now I don´t get display the APP code to register by microsoft a second time?
The led always blinking purple and getting this error:

`23:43:45]
[23:43:50]pollForToken()
[23:43:51][HTTPS] Method: POST, Response code: 400
[23:43:51]pollForToken() - Wating for authorization by user: AADSTS70016: OAuth 2.0 device flow error. Authorization is pending. Continue polling.
[23:43:51]Trace ID: 9d7dc30c-5d95-41ed-b988-693212533400
[23:43:51]Correlation ID: 1705c440-4cd8-479e-85b6-38d4ad179a20
[23:43:51]Timestamp: 2021-01-16 22:43:52Z
[23:43:51]
[23:43:56]pollForToken()
[23:43:58][HTTPS] Method: POST, Response code: 400
[23:43:58]pollForToken() - Wating for authorization by user: AADSTS70016: OAuth 2.0 device flow error. Authorization is pending. Continue polling.
[23:43:58]Trace ID: 953199e7-b7a5-48a6-a5c3-fb88d62c3500
[23:43:58]Correlation ID: 1705c440-4cd8-479e-85b6-38d4ad179a20
[23:43:58]Timestamp: 2021-01-16 22:43:58Z
[23:43:58]
`
greet Lars

toblum, 2021-01-17:
Hi @Eisbaeeer,

I'm not cempletely sure what happened with the app on your side.
Maybe it's the best to register your own app, as described here: https://github.com/AzureAD/microsoft-authentication-library-for-dotnet/wiki/Device-Code-Flow#application-registration and use this client id. Then you have full contrill over the app in Azure Portal.

Greetings

Eisbaeeer, 2021-01-17:
Hmm, I have no rights to create an App in Azure.Von meinem/meiner Galaxy gesendet

Eisbaeeer, 2021-01-17:
How can I get the registration code again? no popup in config, no code will be shown. I only get this:

` pollForToken()
[HTTPS] Method: POST, Response code: 400
pollForToken() - Wating for authorization by user: AADSTS70016: OAuth 2.0 device flow error. Authorization is pending. Continue polling.
Trace ID: 200b2d8e-7540-4b9a-a15a-afef241d4700
Correlation ID: aeadfaa8-3cb7-4888-98e9-20ed364e3685
Timestamp: 2021-01-17 11:36:58Z
`

Eisbaeeer, 2021-01-17:
Ok, problem solved! I used the new Edge instead of Firefox and get with Edge the authentication code to register the app in O365 again. It seems, that firefox not working with that. I tested your newest version and all is running fine!
Thanks!

closed 2021-01-17 by Eisbaeeer

toblum, 2021-01-17:
Yes, this could be that Firefox doesn't work as expected. I tested only with chrome and new edge. Will try it out next time and at least document if there are problems.

Greetings