toblum/ESPTeamsPresence #13: Seems to not working on Arduino NodeMCU ESP32

julienzvh, 2021-02-09:
Hi there,

I'm not sure it's the right area, but I tried to follow your guide concerning the implementation fo the firmware unfortunately without success. 

For the config:
- I've an Arduino NodeMCU ESP32-S (WiFi+Bluetooth)
- I've downloaded NodeMCU-PyFlasher-4.0

I can successfully flash the Board with the frimeware.bin:

```
Command: esptool.py --port /dev/cu.usbserial-0001 --baud 115200 --after no_reset write_flash --flash_mode dio 0x00000 /Users/xxx/Downloads/firmware.bin

esptool.py v2.6
Serial port /dev/cu.usbserial-0001
Connecting.......
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 3c:61:05:03:ac:a0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1078672 bytes to 613160...
Wrote 1078672 bytes (613160 compressed) at 0x00000000 in 54.3 seconds (effective 158.8 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.
```
**After that, I don't get any new WiFi hotspot "ESPTeamsPresence".**

I've tried the 3 different options in the software which are: Quad I/O (QIO), Dual I/O (DIO) and DUAL Output (DOUT)

[image]

Is there any chance that the Board doesn't works with your Frimware?

toblum, 2021-02-09:
Hi @julienzvh,

this looks perfectly fine. DIO should be the right choice.

Some questions:
- Do you have a link to the board you're using? 
- Do you have some experience with the ESP32 or Arduino and have a look at the serial monitor. That would help.
- Maybe you could try to flash a earlier release of the software, maybe https://github.com/toblum/ESPTeamsPresence/releases/tag/v0.16.0, just to make sure it's not related to the latest version.
- Or you could use https://github.com/esphome/esphome-flasher/releases/tag/v1.3.0 to flash the firmware. There is also a "View logs" option that shows you the output of the board.

Greetings
Tobias

julienzvh, 2021-02-09:
Hi Tobias, 

Thank you for your help!

I'm using the following board: https://www.bastelgarage.ch/nodemcu-32s-esp32-wifi-bluetooth-entwicklungs-board
According to the serial monitor, I'm having the following issue:

```
21:28:48.730 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:28:48.730 -> flash read err, 1000
21:28:48.764 -> ets_main.c 371 
21:28:49.110 -> ets Jun  8 2016 00:22:57
```
I will hav a try with the ESPHOME option

Only for my understanding, In the project [here ](https://toblum.github.io/ESPTeamsPresence/#/build/hardware) you mention to connect the D12 but your illustration is D13. Do we need to plug 13 or 12?

julienzvh, 2021-02-09:
Finally, it's worked now with all the setup till I figured out that my organisation doesn't allow to connect Azur App... I will try an alternative with AHK...

toblum, 2021-02-09:
Hi,

the chip looks like a standard ESP32 to me, so no problem here. But I don't know this exact board as there are so many versions out there. Some might be a bit ... special.

From the error message this looks like a problem with a not completely flashed firmware. It can find the bootloader, if I'm right. Maybe using another flash software like the ESPHOME-flasher or [Nodemcu FLasher](https://github.com/nodemcu/nodemcu-flasher) could help.

If that doesn't fix the problem I would try to flash another well known software like [WLED](https://github.com/Aircoookie/WLED/releases/tag/v0.11.1) to that board just to see if the board works in general.

While trying everything out, I would also recommend to leave the LEDs unconnected, to make sure that this is no problem.

PIN D13 should be the right one, P13 on your board. Thank you for pointing me to this. I corrected it.

Let me know if you got it working.

Greetings

toblum, 2021-02-09:
OK, good to know!

It's a pity that this isn't allowed at your org.

You're talking about https://github.com/ajobbins/AHK-Teams-Presence ?
Maybe you could still use the hardware, flash WLED to it to have a HTTP controllable light that you could trigger from AHK...
Just as an idea.

Greetings

closed 2021-02-09 by toblum

jkutianski, 2021-04-22:
I have the same problem when I flash it with https://github.com/marcelstoer/nodemcu-pyflasher, but works fine with https://github.com/esphome/esphome-flasher/releases/tag/v1.3.0