toblum/ESPTeamsPresence #15: No code showing in popup

ThePaulius, 2021-03-19:
Tried all browsers, and I can not get the code to appear, the input is empty, I even looked in the page source, nothing :

"It should say "No authentication infos found, start device login flow to complete widget setup!". Click "Start device login". The LEDs should show now a running purple animation.

A popup opens up

Copy the code that is displayed and click "Open device login" or open https://microsoft.com/devicelogin  in your favorite browser."

MegaWatt71, 2021-03-19:
Hi Paulius! Question: Is there a pop up, but it shows no code or is there no popup at all, when you click on "start device login" ?

If there is no popup at all, then you have an issue with your security settings. For me Opera and Edge work both well.

If you do not see the code in the pop up window, then there might be already an authentication running. 
In this case reset the ESP32 and try again.

ThePaulius, 2021-03-20:
I see the popup in all but FF, and in all cases there is no code... I also watch the logs in the flash tool to make sure its all connected to WiFi etc... and see it's using the right I'd too..  all pretty cool though. Also is the code avaliable as I'd like to just turn a red led on as don't need colours.

ThePaulius, 2021-03-20:
I just tried again this morning, and it worked. Must have confused something and it's all had time to clear out over night.

ThePaulius, 2021-03-26:
It's doing it again now... I'm starting a fresh install, and no browsers showing a code inside the popup. the pop is loading ok.

I wonder if it's being blocked from the Azzure end?  tired this morning, and left it for a few hours, tried again and still not showing a code.

closed 2021-03-26 by ThePaulius