toblum/ESPTeamsPresence #16: invalid header: 0x6c63206e

ThePaulius, 2021-03-24:
Tried to flash f/w to 1.18 and appear to have bricked it!

I previously was getting the lights working, but would not get the Presence, the logs showed it failing, then saw an issue post with a similar problem, so thought id flash and go.

I've tried a few times to revert to the previous f/w also setting 'yes wipes all data in NodeMCU PyFlasher.

oh, dear..

[19:02:56]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]ets Jul 29 2019 12:21:46
[19:02:56]
[19:02:56]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]ets Jul 29 2019 12:21:46
[19:02:56]
[19:02:56]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:56]invalid header: 0x6c63206e
[19:02:57]invalid header: 0x6c63206e
[19:02:57]invalid header: 0x6c63206e
[19:02:57]ets Jul 29 2019 12:21:46
[19:02:57]
[19:02:57]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:02:57]
Serial port closed!

if i run an example sketch from Arduino IDE, like get chip info, that works..  

Have also tried the esp32 reset flash

So annoying, as it was almost working !

toblum, 2021-03-24:
Hi @ThePaulius,

this is strange, didn't see this before. If the other demo sketches from Arduino IDE like "blink" work, then the board itself should be fine. 

Did you try it without anything connected to the ESP32? Sometimes peripherals like the LEDs could irritate the bootloader.

Maybe try this, to get the complete flash of the ESP reset: http://iot-bits.com/erase-esp32-flash-completely/

Greetings

ThePaulius, 2021-03-24:
Hey,

Ok ill try flashing without the leds connected. 

I ran a quick test with a Neopixel example sketch, and that all worked perfectly, lovely rainbow disco lol, so esp32 is still alive and kicking.

ThePaulius, 2021-03-24:
Flashed with nothing connected to it, completed as expected (just as it does when something is connected) but still get the same result, this loop...

MegaWatt71, 2021-03-25:
@ThePaulius : You've set up the Arduino IDE, so have you tried to compile and upload the source code as well? Tell me, if you need help to set up your IDE for this project.

ThePaulius, 2021-03-25:
Hi, no I haven't tried to do the ESPTeamsPresence via the arduini idea, I just used that to see if the MCU was fried.

I've only tried tue f/w flash with ESPTP,  I can't see where I would find the code for the IDE method... so can't try it that way..

ThePaulius, 2021-03-25:
Someone suggested to try adding the line yield(); to the inside of all while loops to avoid watchdog resets. @toblum

ThePaulius, 2021-03-26:
OK, I've managed to factory reset the ESP32 by running the erase 3 times (!).  now I've flashed f/w 0.18 and it seems to be back to defaults, so that's good, I can now run through the setup again.

closed 2021-03-26 by ThePaulius