toblum/ESPTeamsPresence #19: Single quote issue?

pduck27, 2021-07-16:
Hi,

I had an issue with the payload response in the last weeks. I seems that it comes with single quotes in the GUID part of the presence-url-part now. I got an "InvalidInput" error during deserialization. I solved it by replacing singlequotes like this  --> https://github.com/pduck27/ESPTeamsPresence/commit/5aaad5b64a4a1655bf5900337c07c836c7b086f2#diff-9f3581dc75caf236e320a285e2cc1523e851b06566dc6310ed38c92cd8b68cf6. 

Don't you had this problem?

toblum, 2021-07-16:
Hi @pduck27,

I didn't notice any problems lately, but will keep an eye on it.

In which response are the single quotes contained? In the reponse from the /me/presence request?
Do you have an example from such a payload? Didn't see this before.

Greetings
Tobias

pduck27, 2021-07-19:
Hi,

I get this as response for presence request:
[image]

I tried to escape the quotes before deserialization but failed. However, the value is not necessary for later proceeding so I just delete the quotes by now using replace("'", "") before deserialization.

toblum, 2021-07-22:
Hi @pduck27,

thank you for the clarification. I had no problems so far, but will keep an eye on it.
The single quotes should be fine for ArduinoJSON anyway. Strange.

Greetings

toblum, 2021-07-25:
Hi @pduck27,

I saw this behaviour also yesterday and asked also at the ArduinoJSON project: https://github.com/bblanchon/ArduinoJson/issues/1614
Today everything works fine again also with single quotes, so it must originate elsewhere.

Greetings

toblum, 2021-07-26:
Hi @pduck27,

I think the case is solved: https://github.com/bblanchon/ArduinoJson/issues/1614#issuecomment-886989394
Could you check if your code contains this line? https://github.com/toblum/ESPTeamsPresence/blob/41d1980819130d8e429647f2bddb53c438c5e422/src/request_handler.h#L39

Greetings!

pduck27, 2021-07-27:
Hi, thank you, that was it! Sorry for not checking the changes since I forked.

toblum, 2021-07-27:
Great! Good to know that it works and not all users are affected.

closed 2021-07-27 by toblum