toblum/ESPTeamsPresence #23: Status times out troubleshooting?

nauticus25, 2022-01-17:
Great project! Got this working on an Adafruit Huzzah32. I'm connected to my company's guest wifi, which works great for a period of time before the status light appears to get stuck and doesn't update. When I reboot the device, it works great again. I've been using it for a week and I think it times out after an hour or two, though I don't know for sure.

I think I have the polling at 30 seconds. Is the timeout likely to be on the Microsoft side, the wifi connection, or something local to the device? What's the best way to troubleshoot this?

toblum, 2022-01-19:
Hello!

That sounds more like a timeout in the guest wifi. I didn'T experience eny throttlich from MS side so far.

For troubeshooting it would be best to leave the device connected to serial console and check the messages when disconnection happens.

Greetings!

nauticus25, 2022-01-26:
Ok, I've finally had a chance to let it sit while connected via the serial console here at home.

Most of the status looks like this:

`Polling presence info ...`
`[HTTPS] Auth token valid for 4289658 s.`
`[HTTPS] Method: GET, Response code: 200`
`setAnimation: 0, 0-12, Mode: 0, Color: 65280, Speed: 3000`
`--> Availability: Available, Activity: Available`

After a little over 30 minutes, it stopped updating and this is the last message:
`Polling presence info ...`
`[HTTPS] Auth token valid for 4289623 s.`
`[HTTPS] Method: GET, Response code: 401`
`[HTTPS] Other HTTP code: 401`
`Response:`

I did a search on that code and it appears to be an authorization error. I'm not sure if it matters, but it turns out I had polling set to 10 seconds. I'm going to bump it up to 30 seconds to see if that changes anything.

nauticus25, 2022-01-26:
Ok, on 30 second polling it got a 401 response after about 90 minutes. Coincidentally, that's 3 times as long to get a 401 response while using a polling that is also 3 times as long (30 seconds vs 10 seconds).

I forgot to mention that this is my home wifi network which as far as I know does not have any timeouts.

toblum, 2022-03-26:
Hi @nauticus25,

unfortunately I had broken my hand and wasn't able to answer here for some weeks. Is this still an issue?
I didn't see this happen before.

Greetings
Tobias

KdinecCZ, 2023-02-20:
Hi @toblum 
i got same issue as @nauticus25 , device login ends up with message "Teams presence device
You have signed in to the Teams presence device application on your device. You may now close this window."

Console end with this:

> 15:03:33.406 -> ======================================================================
> 15:03:42.002 -> handleRoot()
> 15:04:15.794 -> handleStartDevicelogin()
> 15:04:18.235 -> [HTTPS] Method: POST, Response code: 200
> 15:04:18.282 -> setAnimation: 0, 0-1, Mode: 16, Color: 4194432, Speed: 3000
> 15:04:18.282 -> ======================================================================
> 
//few pollForToken() repeating ... 
> 
> 15:04:37.644 -> pollForToken()
> 15:04:39.923 -> [HTTPS] Method: POST, Response code: 400
> 15:04:39.923 -> pollForToken() - Wating for authorization by user: AADSTS70016: OAuth 2.0 device flow error. Authorization is pending. Continue polling.
> 15:04:39.970 -> Trace ID: 33593875-6bdb-4f73-b792-8298ca252b00
> 15:04:39.970 -> Correlation ID: c0624a79-42a6-4679-a2d2-a0f0c7baaaae
> 15:04:39.970 -> Timestamp: 2023-02-20 14:04:40Z
> 15:04:39.970 -> 
> 15:04:44.965 -> pollForToken()
> 15:04:47.776 -> [HTTPS] Method: POST, Response code: 200
> 15:04:48.098 -> saveContext() - Success: 5009
> 15:04:48.098 -> Polling presence info ...
> 15:04:48.098 -> [HTTPS] Auth token valid for 3624 s.
> 15:04:49.456 -> [HTTPS] Method: GET, Response code: 401
> 15:04:49.456 -> [HTTPS] Other HTTP code: 401
> 15:04:49.456 -> Response: 

Thats all ... i use you azAPP so client ID is same as in guide anad tenant id comes from AZURE ACTIVE DIRECTORY -> PROPERTIES -> Tenant ID ...., after device login attemp i can see this app listed in my azure portal. Can you help ? ...using prebuild firmware.bin in your release page and hw is "LILYGO TTGO T-Internet-POE ESP32-WROOM LAN8720A" )

After restart, output is simillar with same ending..

> 15:35:09.808 -> ======================================================================
> 15:35:09.855 -> WiFi connected
> 15:35:09.855 -> IP address: 192.168.1.74
> 15:35:09.855 -> State changing from: 3 to 4
> 15:35:09.855 -> Accepting connection
> 15:35:09.855 -> State changed from: 3 to 4
> 15:35:09.855 -> setAnimation: 0, 0-1, Mode: 16, Color: 65280, Speed: 3000
> 15:35:09.855 -> startMDNS()
> 15:35:09.855 -> mDNS responder started: ESPTeamsPresence.local
> 15:35:09.901 -> loadContext() - Success
> 15:35:09.901 -> loadContext() - Next: Refresh token.
> 15:35:09.949 -> Wifi connected, waiting for requests ...
> 15:35:09.949 -> setAnimation: 0, 0-1, Mode: 16, Color: 16711680, Speed: 3000
> 15:35:09.949 -> refreshToken()
> 15:35:12.872 -> [HTTPS] Method: POST, Response code: 200
> 15:35:12.872 -> refreshToken() - Success
> 15:35:13.014 -> saveContext() - Success: 5009
> 15:35:13.014 -> ======================================================================
> 15:35:13.014 -> Polling presence info ...
> 15:35:13.060 -> [HTTPS] Auth token valid for 4784 s.
> 15:35:14.328 -> [HTTPS] Method: GET, Response code: 401
> 15:35:14.328 -> [HTTPS] Other HTTP code: 401
> 15:35:14.328 -> Response:

toblum, 2023-03-02:
@KdinecCZ I tried to reproduce the issue, but in my case it works still flawlessly.
From the logs it looks like the device was able to get a token but the token isn't sufficient to get presence info.

Do you have access to the "Enterprisee applications" in AAD. This should look like in the screenshot:
[image]

In the Enterprise app you also have the "Sign-in log". Are there problems listed?

toblum, 2023-03-19:
@KdinecCZ There was a problem with certificates, that is now fixed in [v.0.18.3](https://github.com/toblum/ESPTeamsPresence/releases/tag/v0.18.3)

closed 2023-03-19 by toblum