toblum/ESPTeamsPresence #25: No login Code

DannyEss, 2022-03-04:
Hey, I want to build my Presence Light but I can't get a code in "Start device login". 

[image]

toblum, 2022-03-26:
Hi @DannyEss,

unfortunately I wasn't able to answer for some weeks because I broke my hand. Is this still an issue? 

To see what happens, it would be great to have a look at the serial console of the device when it's plugged in to a PC.

Greetings
Tobias

brian-inksmith, 2022-04-29:
@toblum I am getting the same thing as @DannyEss 
On the console it says 
`[HTTPS] Connection refused`

The fix was to comment the disablecertcheck
`//#define DISABLECERTCHECK 1 `

riraosan, 2022-08-09:
Hi @DannyEss 

Here is my log, I am not able to get the code because I am not able to communicate over SSL due to lack of SRAM.
Are you in the same situation as me?
If there is only about 90KB of heap memory left when the program is executed, the SSL communication process will fail.

[image]
[image]

icaputo, 2022-12-02:
I had the same issue, fixed using the default client ID (and not my own)
Client-ID:  3837bbf0-30fb-47ad-bce8-f460ba9880c3

fabianocattaneo, 2023-03-02:
I'm having the same issue (no code) even if I am using the default cliend ID.

Any suggestion?

toblum, 2023-03-02:
@fabianocattaneo Do you have access to the device console? Or the Azure AD Enterprise apps? This would be helpful.

Also please have a look here: https://github.com/toblum/ESPTeamsPresence/issues/23#issuecomment-1452553793

fabianocattaneo, 2023-03-03:
Indeed I really don't know what happened, a recompile and reflash fixed the issue and it is working ok right now.

toblum, 2023-03-03:
@fabianocattaneo Great! Sometimes it just happens that way with software. :-)

closed 2023-03-03 by toblum