toblum/ESPTeamsPresence #27: Do you have a plan to library the AzuruAD login process?

riraosan, 2022-08-03:
HI. @toblum 

I'm @riraosan. 

I have used your app and I am very impressed with the endless possibilities. I am impressed with the endless possibilities of your app. Thank you for your contribution.

I would like to incorporate your work into my project. I am thinking that I can make a library of the login process part to AzuruAD that you have created.

Before I do that, I have a question: have you already realized the libraryization of the AzuruAD login process?
Or do you have a plan to library this AzuruAD login process?

Best regards.

toblum, 2022-08-05:
Hi @riraosan,

I indeed started to build a library out of the authentication and Graph stuff. It's here: https://github.com/toblum/ArduinoMSGraph
Also published to platformio: https://registry.platformio.org/libraries/toblum/ArduinoMSGraph

But I never finished it, due to missing time. And I'm not experienced in building Arduino libraries. That was my first try and not the best approach.

Feel free to use and enhance what I already have. 

There is also a small project (diplays your agenda on a epaper display) that uses the library: https://github.com/toblum/ESP32GraphGadget

Greetings
Tobias

closed 2022-08-05 by toblum

riraosan, 2022-08-08:
Hi. @toblum 

Thanks for your reply. I thought it was a very interesting initiative.
As long as you commit to this, I am sure someone somewhere, including me, will modify it in some way.
Thanks.

toblum, 2022-08-08:
I'm totally fine if someone builds a solution based on this. It's all MPL licence. 

Let me know if you make any progress.

riraosan, 2022-08-17:
返信、ありがとうございます。

まだ成果は無いですが、私はあなたのソースコードの解析を始めました。私はプレゼンス以外のGraphAPIを叩く簡単なライブラリを作ることとしました。
例えば、チャンネルに投稿されたメッセージのタイトルを表示する、とか。（私は私のオフィスで使いたいと思っています。私は物理的な光でデプロイをチームメンバーへ通知したいのです。）
また、今後、あなたと相談させてください。
よろしくお願いします。

riraosan, 2022-08-24:
@toblum 

I started organizing my files to make them into a library.

https://github.com/riraosan/ESPTeamsPresence/tree/develop/src

toblum, 2022-08-24:
Nice build! The M5StackAtom with Matrix is a great device to use for this purpose.

Greetings!

riraosan, 2022-08-30:
Hi. @toblum 

I refactored your source code and encapsulated it into a class. There are some minor glitches, but I am now able to connect to WiFi and log in to AzuruAD via the normal path. I read the source code and I appreciate your idea. Thank you very much.

I'd like to send pull-request this source code for you, but I'm not sure if you need it.

If not, may I create an alias repository in my repository and inherit the development as my project?
I would be happy to interact with you.

toblum, 2022-08-30:
Hi @riraosan,

good to hear that you made it work. 

I appreciate that you want to send a pull request, but I think it wouldn't fit in the repo here. So I would just be happy if you just mention in your repo that your work is based on this project as a source.

Many greetings

riraosan, 2022-08-30:
Hi. @toblum 

I would like to sublimate your results into a different form.
In doing so, I would like to show in my repository that I stand on the shoulders of your source code.
https://github.com/riraosan/ESPTeamsPresence/tree/develop

Your ideas are the best.👍　Thanks for your contribution.

toblum, 2022-10-04:
I'll definitely have a look at it. 
Have it working on the M5Stack means that it's easy to try out and modify, even for beginners.