toblum/ESPTeamsPresence #28: Authentication fails

Laess3r, 2022-09-03:
Hy!
I created a new app in azure to get my own client ID.

When trying to log in my user account using the "Start device Logon" an error is thrown. 
The microsoft login sceen says "login successful", but the device log shows an error:

https://login.microsoftonline.com/error?code=7000218

`[17:01:12]mDNS responder started: ESPTeamsPresence.local
[17:01:12]loadContext() - File empty
[17:01:12]Wifi connected, waiting for requests ...
[17:01:12]======================================================================
[17:01:18]handleRoot()
[17:01:29]handleStartDevicelogin()
[17:01:34][HTTPS] Method: POST, Response code: 200
[17:01:34]setAnimation: 0, 0-16, Mode: 16, Color: 4194432, Speed: 3000
[17:01:34]======================================================================
[17:01:39]pollForToken()
[17:01:43][HTTPS] Method: POST, Response code: 400
[17:01:43]pollForToken() - Wating for authorization by user: AADSTS70016: OAuth 2.0 device flow error. Authorization is pending. Continue polling.
[17:01:43]Trace ID: 161eeca5-3b0e-4d84-b776-4bf8158e5d00
[17:01:43]Correlation ID: e1fb7471-56eb-4481-942b-603f1007f764
[17:01:43]Timestamp: 2022-09-03 15:01:29Z
[17:01:43]
[17:01:48]pollForToken()
[17:01:52][HTTPS] Method: POST, Response code: 401
[17:01:52][HTTPS] Other HTTP code: 401
[17:01:52]Response: {"error":"invalid_client","error_description":"AADSTS7000218: The request body must contain the following parameter: 'client_assertion' or 'client_secret'.\r\nTrace ID: b0376132-d7d0-4757-8705-6bbca5677800\r\nCorrelation ID: e1fb7471-56eb-4481-942b-603f1007f764\r\nTimestamp: 2022-09-03 15:01:38Z","error_codes":[7000218],"timestamp":"2022-09-03 15:01:38Z","trace_id":"b0376132-d7d0-4757-8705-6bbca5677800","correlation_id":"e1fb7471-56eb-4481-942b-603f1007f764","error_uri":"https://login.microsoftonline.com/error?code=7000218"}
[17:01:52]Device login failed
[17:01:52]======================================================================`

Do you have any idea how to fix this? Thank you!

Laess3r, 2022-09-03:
By setting the azure account to "multi instance" i got one step further. 
But it still seems to get a 401 trying to get my teams status?

`[17:09:39]======================================================================
[17:09:39]WiFi connected
[17:09:39]IP address: 192.168.0.104
[17:09:39]State changing from: 3 to 4
[17:09:39]Accepting connection
[17:09:39]State changed from: 3 to 4
[17:09:40]setAnimation: 0, 0-16, Mode: 16, Color: 65280, Speed: 3000
[17:09:40]startMDNS()
[17:09:40]mDNS responder started: ESPTeamsPresence.local
[17:09:40]loadContext() - Success
[17:09:40]loadContext() - Next: Refresh token.
[17:09:40]Wifi connected, waiting for requests ...
[17:09:40]setAnimation: 0, 0-16, Mode: 16, Color: 16711680, Speed: 3000
[17:09:40]refreshToken()
[17:09:44][HTTPS] Method: POST, Response code: 200
[17:09:44]refreshToken() - Success
[17:09:44]saveContext() - Success: 4880
[17:09:44]======================================================================
[17:09:44]Polling presence info ...
[17:09:44][HTTPS] Auth token valid for 4572 s.
[17:09:46][HTTPS] Method: GET, Response code: 401
[17:09:46][HTTPS] Other HTTP code: 401`

toblum, 2022-09-03:
Hi @Laess3r,

the stock app is setup this way:
[image]
You may not need to set it to multitenant if you're using it just on the same tenant.

Permissions are set like this:
[image]

That works for the stock app registration.,

Greetings

thechrispino, 2023-01-03:
Any Update on this 401 error? Can you provide the details of the app you created inside azure so that I can test replicating it within my subscription?

toblum, 2023-01-04:
Hi @thechrispino,

this won't work with consumer accounts indeed. 
I'm not actively working on this now.
There is a recent library version of this, maybe you want to have a look at it: https://github.com/toblum/ESPTeamsPresence/issues/27

Greetings

closed 2023-01-04 by toblum