toblum/ESPTeamsPresence #29: SSL connection errors (fix included ;) )

NochEinKamel, 2023-03-10:
I just stumbled upon this gem, but couldn't really get it to work.

The normal firmware.bin was working for the device authentication flow but had certificate errors when polling the actual status;
the nocertcheck-version wasn't working at all (ssl connection error "-1"...).

After digging into it a bit I discovered that the software uses two different root certificates to validate the requests to https://login.microsoftonline.com (DigiCert Global Root CA) and to https://graph.microsoft.com (DigiCert Assured ID Root G2).

But upon inspecting the certificates used by those two domains (using https://graph.microsoft.com/.default) I realized that both are using the DigiCert Global Root CA and the graphql endpoint was no longer using the G2 root certificate.

So I did what any sane developer would do and just commented out some lines:

In request_handler.h

	//if (url.indexOf("graph.microsoft.com") > -1) {
	//	client->setCACert(rootCACertificateGraph);
	//} else {
		client->setCACert(rootCACertificateLogin);
	//}

That way all requests use the Root CA and everything works.

nerazifel, 2023-03-12:
> I just stumbled upon this gem, but couldn't really get it to work.
>
> The normal firmware.bin was working for the device authentication flow but had certificate errors when polling the actual status; [...]

Hi @NochEinKamel , did you generate your own Firmware? I have found the rquest_handler.h file and commented out the line, but am struggling with uploading it to the Arduino? Can you share the file?

Thanks

NochEinKamel, 2023-03-13:
@nerazifel Sorry, I'm a noob regarding those things. I just pushed the firmware directly from vs code with platformio plugin onto the board and have no idea how to generate a firmware file.

jbetsinger, 2023-03-14:
@NochEinKamel this worked for me.

On line 40 of the main.cpp file there is a link for a "Tool to get certs: https://projects.petrucci.ch/esp32/".  Open this link and then copy each of the "https://login.microsoftonline.com" and "https://graph.microsoft.com" links onto this page.  Press "Get Certificate" then select the "DigiCert SHA2 Secure Server CA" cert by clicking on the certificate icon.  Then scroll to the bottom and copy the "ESP32 code:" and replace the code sections within the main.cpp file.  Take note of the variable names prior to pasting then modify them back to their original names.

toblum, 2023-03-19:
Hi @NochEinKamel and @jbetsinger, 

thanks for bringing the problem to my attention. It worked for me until recently, now I could also see and reproduce the problem. Apparently the certificate changes are currently being rolled out gradually.

As you already mentioned: Both endpoints are now apparently using the same root certificate. I have now included the latest certificate and applied it to both endpoints.

The latest release is now [v.0.18.3](https://github.com/toblum/ESPTeamsPresence/releases/tag/v0.18.3). I have also had to update some of the libraries used.
@nerazifel If you like, please feel free to try the release.

Many greetings
Tobias

closed 2023-03-19 by toblum