toblum/ESPTeamsPresence #33: LED's do not turn on.

westleyp, 2023-07-02:
Board: Adafruit ESP32-S2 Feather.

I get expected result via serial monitor:
```Polling presence info ...
[HTTPS] Auth token valid for 3905 s.
[HTTPS] Method: GET, Response code: 200
setAnimation: 0, 0-8, Mode: 0, Color: 65280, Speed: 3000
--> Availability: Available, Activity: Available`
```

When I compile WS2812FX and use example:
```
#include <Arduino.h>

#include <WS2812FX.h>

#define LED_COUNT 8
#define LED_PIN 12
#define WHITELEDONLY (uint32_t)0xFF000000 /* for WHITE GRB LEDs */

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRBW + NEO_KHZ800);

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(100);
  ws2812fx.setSpeed(5000);
  ws2812fx.setColor(WHITELEDONLY);
  ws2812fx.setMode(0);
  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}
```
Neopixel strip lights up as expected.
Is there something different I should be doing for the Adafruit ESP32-S2 Feather?

platformis.ini
```
[platformio]
src_dir = ./src/
default_envs = featheresp32-s2
description = The Microsoft Teams Neopixel Presence Device for ESP32

[env]
platform=espressif32
board=esp32dev
framework=arduino
; upload_port=COM4
monitor_speed=115200
upload_speed=921600
build_flags=
    -DDATAPIN=13
    -DNUMLEDS=16
    ; -DCORE_DEBUG_LEVEL=5
lib_deps=
  IotWebConf@2.3.3
  ArduinoJson@6.21.0
  WS2812FX@1.4.1

[env:esp32doit-devkit-v1]
board=esp32doit-devkit-v1

[env:esp32doit-devkit-v1-nocertcheck]
board=esp32doit-devkit-v1
build_flags=
    ${env.build_flags}
    -DDISABLECERTCHECK
  
[env:m5stack-core-esp32]
platform=espressif32
extends=esp32dev
board=m5stack-core-esp32
upload_speed=115200
build_flags=
    -DDATAPIN=26
    -DNUMLEDS=37

[env:featheresp32-s2]
platform = espressif32
board = featheresp32-s2
framework = arduino
build_flags=
    -DDATAPIN=12
    -DNUMLEDS=8
```

toblum, 2023-07-02:
Hi @westleyp,

unfortunately, I don't have experiences with the S2 or other ESP32 variations. It's build and tested for the standard ESP32.
The main issue might be that the S2 has only one core, while the standard ESP32 has two and both are used in the code, one for the WiFi communication, one for the LEDs. So I'm wondering why it's running at all. :-/
You might be able to rewrite the code and let everything run on the same core, but this might lead to LED flickering when WiFi communication is active.

Greetings

westleyp, 2023-07-02:
Hey @toblum 

Thanks for the quick response. I'll do some messing around, If I get it to work I'll give you a heads up.

Thanks Again!

closed 2024-01-12 by toblum