toblum/ESPTeamsPresence #35: No device Login code

NKaaay, 2023-08-16:
Hey Tobias,

i want to build also a Teams Presence Light like yours but the firmware did not give me a Login code.
I hope you could help me there.
Here a screenshot:
[image]
testet with firmware 0.18.1 & 0.18.3

Best regards
Niclas

toblum, 2023-08-16:
Hi @NKaaay,

that could be a wrong ClientID and/or tenant name. You should double-check these both values, also for leading/trailing spaces.
If these are correct then a look at the serial console of the ESP might help. And errors shown there?

Greetings!

NKaaay, 2023-08-17:
Ok Thanks. I have solved this problem with my Tenant GUID instead of the eMail-Adress.
But now i have an other problem. I have no admin rights in the organization and can not complete the last step...

toblum, 2023-08-17:
OK, good to know that it works in general. But you'll need someone with appropriate rights to approve that last step if it's restricted in your organization. No way around that.

closed 2023-08-17 by toblum