toblum/ESPTeamsPresence #36: Set leds offline when the hour is non-workable

rodritwm, 2023-09-02:
hello is it possible to add a little function that gets the time and if the time is no between 9am and 18hs the leds are offline? Thanks!

toblum, 2023-09-02:
Hi @rodritwm,

time controlled functions wouldn't be too easy because the ESP isn't aware of the time and has no built-in clock.

What do you want to accomplish? Setting your status to on-/offline at this times? This might be something you could accomplish using PowerAutomate. 
If you just want to turn the LEDs off, I would go with a simple time-controlled socket, because that would save also power.

Greetings
Tobias

bertenvdb, 2024-09-04:
Alternativly, you could get the time by doing a GET request on e.g. `http://worldtimeapi.org/api/ip`.