toblum/ESPTeamsPresence #38: No Bootable Partition?

kirktindel, 2023-09-20:
I feel sheepish asking here, but I've been banging my head against the wall for 5 hours last night and 4 hours today.  I've researched the errors I've received and I'm getting all sorts of different answers.

The issue is constant - after flashing firmware, the program won't boot.  Here is a log of what I'm seeing.  I've included the upload.

```
esptool.py v2.6-beta1
Serial port COM7
Connecting...
.
.
.
.

Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: b0:b2:1c:a8:5f:14
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1100256 bytes to 702956...

Writing at 0x00010000... (2 %)

Writing at 0x00014000... (4 %)

Writing at 0x00018000... (6 %)

Writing at 0x0001c000... (9 %)

Writing at 0x00020000... (11 %)

Writing at 0x00024000... (13 %)

Writing at 0x00028000... (16 %)

Writing at 0x0002c000... (18 %)

Writing at 0x00030000... (20 %)

Writing at 0x00034000... (23 %)

Writing at 0x00038000... (25 %)

Writing at 0x0003c000... (27 %)

Writing at 0x00040000... (30 %)

Writing at 0x00044000... (32 %)

Writing at 0x00048000... (34 %)

Writing at 0x0004c000... (37 %)

Writing at 0x00050000... (39 %)

Writing at 0x00054000... (41 %)

Writing at 0x00058000... (44 %)

Writing at 0x0005c000... (46 %)

Writing at 0x00060000... (48 %)

Writing at 0x00064000... (51 %)

Writing at 0x00068000... (53 %)

Writing at 0x0006c000... (55 %)

Writing at 0x00070000... (58 %)

Writing at 0x00074000... (60 %)

Writing at 0x00078000... (62 %)

Writing at 0x0007c000... (65 %)

Writing at 0x00080000... (67 %)

Writing at 0x00084000... (69 %)

Writing at 0x00088000... (72 %)

Writing at 0x0008c000... (74 %)

Writing at 0x00090000... (76 %)

Writing at 0x00094000... (79 %)

Writing at 0x00098000... (81 %)

Writing at 0x0009c000... (83 %)

Writing at 0x000a0000... (86 %)

Writing at 0x000a4000... (88 %)

Writing at 0x000a8000... (90 %)

Writing at 0x000ac000... (93 %)

Writing at 0x000b0000... (95 %)

Writing at 0x000b4000... (97 %)

Writing at 0x000b8000... (100 %)

Wrote 1100256 bytes (702956 compressed) at 0x00010000 in 9.0 seconds (effective 979.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Success!
e
ts Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5564
load:0x40078000,len:0
load:0x40078000,len:13756
entry 0x40078fb4
[0;32mI (29) boot: ESP-IDF v3.0.3 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 08:53:32[0m
[0;32mI (30) boot: Enabling RNG early entropy source...[0m
[0;32mI (34) boot:
 SPI Speed      : 40MHz[0m
[0;32mI (38) boot: SPI Mode       : DIO[0m
[0;32mI (42) boot: SPI Flash Size : 4MB[0m
[0;32mI (46) boot: Partition Table:[0m
[0;32mI (49) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (57) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (64) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (72) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32m
I (79) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (87) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (94) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (102) boot: End of partition table[0m
[0;31mE (106) boot: ota data partition invalid and no factory, will try all partitions[0m
[0;31mE (114) esp_image: image at 0x100000 has invalid magic byte[0m
[0;33mW (120) esp_image: image at 0x100000 has
 invalid SPI mode 8[0m
[0;33mW (126) esp_image: image at 0x100000 has invalid SPI size 15[0m
[0;31mE (133) boot: OTA app partition slot 0 is not bootable[0m
[0;31mE (138) esp_image: image at 0x280000 has invalid magic byte[0m
[0;33mW (145) esp_image: image at 0x280000 has invalid SPI mode 255[0m
[0;33mW (151) esp_image: image at 0x280000 has invalid SPI size 15[0m
[0;31mE (157) boot: OTA app partition slot 1 is not bootable[0m
[0;31mE (163) boot: No bootable app partitions in the partit
ion table[0m
user code done

```

I have used the Home Flasher and the PyQt5 Flasher.  I have used two different Windows computers, 5 different ESP32 boards and those boards come from three different batches of boards I've purchased before.  I've tried holding down the boot button the entire time the firmware loads, or using a 10 ohm cap on one board to get it to go into boot mode.  

No matter what happens, every time this thing boots up after downloading the firmware, it keeps basically no doing anything.  I tried re-downloading the firmware.bin and I also tried replacing it with the no cert firmware file.  All always result in the same thing.  I also have used two different USB cables (3 of the 5 ESP boards are USB-C, two are Micro USB) and I even tried using an FTDI cable, all to no avail.

Again, I feel foolish asking here, because I am pretty sure there is something fundamental I am missing.  As we say in the IT industry "the root cause was between the keyboard and the chair".  Does anyone know what newbie mistake it is that I'm making here?  I am at my wits end. I can show you the pile of ESPs on my bench right now, it's shameful.  I've pulled known working good ones that I've used for other projects, but nothing works.

kirktindel, 2023-09-20:
Oh, and I've researched each of those listed errors (and gotten a different answer for each), but I felt it pertinent to mention that most of the boards I'm using are brand new out of the package boards.  I've used the flasher before for other projects and never had an issue, although it's been a couple of years as I mostly do Raspberry Pi stuff now.  Since I don't see anyone else having this same issue, I'm assuming the problem is with something I'm doing.  Perhaps I'm downloading the firmware.bin file incorrectly?  I've been just right-clicking and save as.

Anyway, thanks again and sorry to keep bothering you guys with my dumb questions.  I'm trying to make a bunch of these for the folks on my team.  Great project!

kirktindel, 2023-09-21:
I had a friend who lives across the country try to flash his, now he has the exact same issue.  Same error messages, same everything.  What could we be doing wrong?

toblum, 2023-09-21:
Hi @kirktindel,

this is really strange. I did not experience this before. I could not try it myself atm. Could you try with an older release, e.g. https://github.com/toblum/ESPTeamsPresence/releases/tag/v0.18.0 Just to make sure that the latest build isn't broken.

Greetings
Tobias

kirktindel, 2023-09-22:
Unfortunately, the exact same thing happens.  The problem HAS to be me, but I can't figure out what it is.  Here's the output:

```
esptool.py v2.6-beta1
Serial port COM7
Connecting...
.
.
.
.
.
_

Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: b0:b2:1c:a8:74:50
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1084256 bytes to 629462...

Writing at 0x00010000... (2 %)

Writing at 0x00014000... (5 %)

Writing at 0x00018000... (7 %)

Writing at 0x0001c000... (10 %)

Writing at 0x00020000... (12 %)

Writing at 0x00024000... (15 %)

Writing at 0x00028000... (17 %)

Writing at 0x0002c000... (20 %)

Writing at 0x00030000... (23 %)

Writing at 0x00034000... (25 %)

Writing at 0x00038000... (28 %)

Writing at 0x0003c000... (30 %)

Writing at 0x00040000... (33 %)

Writing at 0x00044000... (35 %)

Writing at 0x00048000... (38 %)

Writing at 0x0004c000... (41 %)

Writing at 0x00050000... (43 %)

Writing at 0x00054000... (46 %)

Writing at 0x00058000... (48 %)

Writing at 0x0005c000... (51 %)

Writing at 0x00060000... (53 %)

Writing at 0x00064000... (56 %)

Writing at 0x00068000... (58 %)

Writing at 0x0006c000... (61 %)

Writing at 0x00070000... (64 %)

Writing at 0x00074000... (66 %)

Writing at 0x00078000... (69 %)

Writing at 0x0007c000... (71 %)

Writing at 0x00080000... (74 %)

Writing at 0x00084000... (76 %)

Writing at 0x00088000... (79 %)

Writing at 0x0008c000... (82 %)

Writing at 0x00090000... (84 %)

Writing at 0x00094000... (87 %)

Writing at 0x00098000... (89 %)

Writing at 0x0009c000... (92 %)

Writing at 0x000a0000... (94 %)

Writing at 0x000a4000... (97 %)

Writing at 0x000a8000... (100 %)

Wrote 1084256 bytes (629462 compressed) at 0x00010000 in 8.4 seconds (effective 1027.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Success!
"%=m[0m
[0;32mI (42) b
oot: SPI Flash Size : 4MB[0m
[0;32mI (46) boot: Partition Table:[0m
[0;32mI (49) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (57) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (64) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (72) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (79) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (8
7) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (94) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (102) boot: End of partition table[0m
[0;31mE (106) boot: ota data partition invalid and no factory, will try all partitions[0m
[0;31mE (114) esp_image: image at 0x100000 has invalid magic byte[0m
[0;33mW (120) esp_image: image at 0x100000 has invalid SPI mode 38[0m
[0;31mE (126) boot: OTA app partition slot 0 is not bootable[0
m
[0;31mE (132) esp_image: image at 0x280000 has invalid magic byte[0m
[0;33mW (138) esp_image: image at 0x280000 has invalid SPI mode 255[0m
[0;33mW (145) esp_image: image at 0x280000 has invalid SPI size 15[0m
[0;31mE (151) boot: OTA app partition slot 1 is not bootable[0m
[0;31mE (157) boot: No bootable app partitions in the partition table[0m
user code done

```

To be clear - I:

Run the flasher tool as admin (Windows 11, USB-C to USB-C cable)
I click the firmware button, select the downloaded firmware.bin (and now this new version)
I click the "flash firmware" button, it starts to attempt to load and gives dots or dashes
I press the "boot" button on the ESP for about 2 seconds, then release.  
The firmware appears to load
I wait, then click "open".  Sometimes the program just closes itself, I have to reopen it and then click "open"
I see the same error message.  If I click the RST button on the ESP, the message is repeated.

This is driving me nuts.

kirktindel, 2023-09-22:
You're not going to believe this - the 7th board I tried worked.  I used a new computer (number 3) and new cable (also number 3) and a new board (number 7).

I can swap computers and cables and it still works on this 7th board, so there seems to be a problem with my boards or perhaps voltage somewhere has caused some issue?  I will need to investigate further, but this is definitely a hardware issue on my side.  I apologize for wasting your time again.

closed 2023-09-22 by kirktindel

toblum, 2023-09-22:
@kirktindel This is really strange. I had bad boards before, but only single cases.