toblum/ESPTeamsPresence #40: red animated light

Teeleoffel, 2024-01-09:
Hey toblum

I have an issue with my presenter. All worked before. 
Now i  have the problem that the presenter just show the red animated light. Wifi work. I have acces to the controller. My Teams account should be right (i checked it 3 times. I restarted the process 3 times. I don't know what to do anymore. Is there a solution for my problem ?

Kind regards

Nico

toblum, 2024-01-09:
Hi Nico,

I'm not aware of any changes here.
The Graph API returns different states and these are mapped to different animations here: https://github.com/toblum/ESPTeamsPresence/blob/53d8780fe57f0daa1200a71ba62c1e1518b3cc23/src/main.cpp#L302

According to the [docs](https://learn.microsoft.com/de-de/graph/api/resources/presence?view=graph-rest-beta#properties) these are 15 different states and these are mapped to different animations. Presenting was always the red animation.

Greetings
Tobias

Teeleoffel, 2024-01-09:
Hmm ok. Thats Strange. After the configuration with the api and the Token. It stays in the red Circle Animation. 
I will try to refresh the whole Software and Start from
The beginning. The presence of my Boss that i made hin works without any Problems. 

Thx.

Teeleoffel, 2024-01-11:
Hey Tobias

I found the Solution for my issue. 
When i Write a Note in my presence state, it seams that the device can't find the Right Status to indicate. So the system doesn't work. 
Perhaps later there can be a fix for this issue. 

Kind regards 

Nico

closed 2024-01-12 by toblum

toblum, 2024-01-12:
Hi @Teeleoffel,

this was a good hint. When requesting the presence data, the status note is also included in the response (this wasn't the case when creating the project). When this is longer, the reserved space for the variable reading the response wasn't sufficient. So I reserved more space so that all possible responses should fit not. In my tests it even works with the maximal note size.

There is a new release https://github.com/toblum/ESPTeamsPresence/releases/tag/v0.18.4

Please drop me a note if this fixes the issue for you too.

Many greetings
Tobias

Teeleoffel, 2024-01-12:
I'll try on Monday at work. There is my light. 

Thx for the fast "hopefull" fix. 

Kind regards

Nico

Teeleoffel, 2024-01-15:
Just for the records. It works. I updated to the current Version and it work. 

Thx. 

Nico

toblum, 2024-01-15:
Great! Thank you for the feedback.

commit 198354ee (2024-01-12): Fixes #40: Enlarge DynamicJsonDocument for pollPresence()
to account for status messages in /me/presence response.