toblum/ESPTeamsPresence #42: No code provided when setting up a new device

lohrm-stabl, 2024-05-21:
[v0.18.4](https://github.com/toblum/ESPTeamsPresence/releases/tag/v0.18.4)

This box is empty:
[image]

Tried it with multiple ESP32-WROOM-32.

From the device log:
```
Saving config 'pollInterval'= '30'
Saving config 'numLeds'= '12'
Configuration was updated.
handleRoot()
handleStartDevicelogin()
[ 67666][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -1
[HTTPS] Request failed: connection refused
handleRoot()
handleStartDevicelogin()
[ 75750][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -1
[HTTPS] Request failed: connection refused
handleRoot()
[457861][E][WebServer.cpp:648] _handleRequest(): request handler not found
Requested non-existing page '/favicon.ico' arguments(GET):0
handleFileRead: /favicon.ico
handleStartDevicelogin()
[458103][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -1
[HTTPS] Request failed: connection refused
```

JavaScript console:
```
startDevicelogin 
Object { error: "devicelogin_unknown_response" }
```

lohrm-stabl, 2024-05-22:
Does neither work with generic ID nor custom app.

lohrm-stabl, 2024-05-22:
Seems like this only happens when flashed with the `-nocertcheck` target

toblum, 2024-05-22:
Hi @lohrm-stabl,

I could not look into this so far, because I have no device at hand to test it out on my side.
According to the log it's related to the SSL connection to MS servers. This connections fails and that leads to the described behaviour.
I got mixed results with `--nocertcheck` in the past and this version might be unreliable. Does the normal version work? 

Greetings
Tobias

lohrm-stabl, 2024-05-23:
Hey @toblum,
yes got it to work with the normal version!

toblum, 2024-05-23:
@lohrm-stabl Thank you for the feedback. I might get rid of the nocert version in the future as it seems to cause more problem than it solves. :-)

closed 2024-05-23 by toblum