toblum/ESPTeamsPresence #43: Bug: refreshToken - Error: | 400

lohrm-stabl, 2024-06-24:
Sometimes the light just does not work. Othertimes it works fine. Sometimes it's after a running for 24 hours, sometimes it happens after 5 minutes.

The log then shows: 
```
HTTPS] Method: POST, Response code: 400
refreshToken() - Error:
refreshToken()
[HTTPS] Method: POST, Response code: 400
refreshToken() - Error:
refreshToken()
[HTTPS] Method: POST, Response code: 400
refreshToken() - Error:
```

And the light lights up in red (turning, with 1/4 LEDs on).

toblum, 2024-06-29:
Hi @lohrm-stabl,

I couldn't reproduce this issue so far. Something seems to go wrong when refreshing the token. Unfortunately you can't see details in the logs. So I [published a new version v0.18.5](https://github.com/toblum/ESPTeamsPresence/releases/tag/v0.18.5) that has extended logging in this case. You may try with this version and then we may see why this fails.

Greetings

closed 2024-07-25 by toblum

marcelstoer, 2024-12-11:
Just came across the HTTP 400 as well. The extended logging revealed that it was because I hadn't used the device for >90d. Hence, the refresh token had expired in the meantime.
I had to wipe the flash and setup the app from scratch. Forcefully putting into AP mode and running through the login procedure might have helped as well; didn't try.