toblum/ESPTeamsPresence #45: Crash reboots with current ESP32 platform

marcelstoer, 2024-12-16:
After a long hiatus, I recently "deployed" my device again (see https://github.com/toblum/ESPTeamsPresence/issues/43#issuecomment-2537209161). I did so using a recent ESP32 platform. I immediately noticed that the device would crash reboot every few minutes. The logs didn't reveal anything conclusive yet. However, it looks like the crashes are caused by some invalid state in the HTTP stack.

At the moment I put this up simply as an indicator for other users. If I find the time time I will investigate this a bit.

nmtoblum, 2024-12-18:
Thank you, I may also have a look when I find time.

Many greetings!

toblum, 2024-12-18:
Thank you, I may also have a look when I find time.

Many greetings!

closed 2025-11-09 by toblum