toblum/ESPTeamsPresence #46: No authcode

tophe75, 2025-01-02:
Once starting the "start device login" no auth code is shown
Running on a ESP32-Dev board with a 4x4 led WS2812 matrix
Could it be related to using your generic Client-ID?
[image]

toblum, 2025-01-04:
Hi @tophe75,

I can't test at the moment as I have changed my job and lost my M365 developer sandbox, so I have no non-productive Teams licences to test with. I have to see where I can get some.

But from what I can see, the generic app registration is still active and should work. 

If you can provide the output of the monitor / console of the device, we might get a clue what is wrong.

Many greetings

cschwarz007, 2025-01-12:
I think it may be that @tophe75 you did not enter the client id. Do not leave the field blank, but copy the generic ID into it (that was my error as well).
@toblum : Can you provide me with the Azure App as I may want to set up my own to ensure proper compliance with out corporate standards?

tophe75, 2025-01-12:
@cschwarz007 I have entered the client id provided in the "documentation" and also tried with my own, but I get the same thing, I have not had any time to look in to this any further.

toblum, 2025-01-17:
Hi @cschwarz007,

I documented to steps to setup your own App registration here: https://github.com/toblum/ESPTeamsPresence/issues/30#issuecomment-1475274011

Many greetings!

closed 2025-11-09 by toblum