toblum/ESPTeamsPresence #47: Cannot login device

dpalma2011, 2025-07-07:
When I try to log in to the device, I get this error message. 
I've already checked the settings, the app ID, and the tenant ID. I've also tried the default app ID.

Device Log

======================================================================
handleRoot()
handleStartDevicelogin()
[HTTPS] Method: POST, Response code: 200
                                        setAnimation: 0, 0-16, Mode: 16, Color: 4194432, Speed: 3000
                                                                                                    ======================================================================
pollForToken()
              [HTTPS] Method: POST, Response code: 400
                                                      pollForToken() - Wating for authorization by user: AADSTS70016: OAuth 2.0 device flow error. Authorization is pending. Continue polling. Trace ID: e7927f2f-8ca8-4643-a83f-8c5e67010d00 Correlation ID: 898e99f8-792d-46f0-bde0-bd22a13051d3 Timestamp: 2025-07-07 21:53:08Z

                                                                                     pollForToken()
                                                                                                   [6531207][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():273]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[6531221][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -9984
[HTTPS] Request failed: connection refused
                                          Device login failed
======================================================================

toblum, 2025-07-12:
Hi @dpalma2011,

on the first look it seems that the token poll fails because the SSL certificate could not be validated. This is a issuse because the certificates are baked into the firmware. But from what I could see, the supplied certificates sould still be valid.
MFA for device login should be be a problem in general as far as I remember.

Unfortunately I'm currently not able to test the firmware myself, because I don't have access to a working M365 tenant right now as there are no free developer tenants anymore and a work at a new company where I don't have access to other test tenants. :-(

I'll see if I can find out anything.

Greetings

Slopert1, 2025-11-09:
You're the best. I tried to flash the bin files, but they did not work for me (esphome-flasher and NodeMCU-PyFlasher). They did also not work with v0.18.5.
I build the -nocheckversion on my computer and I could login and see my status changed when I changed it within Teams.
I will no build and check with the cert checks.

toblum, 2025-12-04:
@krisjohnson222  I'm not sure as i don't have a way to test them and i have lost access to them anymore. :-(