toblum/ESPTeamsPresence #48: Successful compilation of platform.io, for ESP32-C3.  Uploaded, nothing happens.

alexburkert, 2025-08-13:
To Whom It May Concern:

As mentioned in the title, I'm trying this project on an ESP32-C3.  The firmware.bin that ships with the project is inappropriate for the ESP32-C3 board.

I've opted to compile for the ESP32-C3 using as a basis the platform.io which ships with the project, modified for the ESP32-C3 board.

Please find it as follows  : 

`;PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
src_dir = ./src/
default_envs = esp32-c3-devkitm-1
description = The Microsoft Teams Neopixel Presence Device for ESP32

[env]
platform=espressif32
board=esp32dev
framework=arduino
; upload_port=COM4
monitor_speed=115200
upload_speed=921600
build_flags=
    -DDATAPIN=13
    -DNUMLEDS=35
    ; -DCORE_DEBUG_LEVEL=5
lib_deps=
  IotWebConf@2.3.3
  ArduinoJson@6.21.0
  WS2812FX@1.4.1

[env:esp32doit-devkit-v1]
board=esp32doit-devkit-v1

[env:esp32-c3-devkitm-1]
board=esp32-c3-devkitm-1

[env:esp32doit-devkit-v1-nocertcheck]
board=esp32doit-devkit-v1
build_flags=
    ${env.build_flags}
    -DDISABLECERTCHECK

[env:m5stack-core-esp32]
platform=espressif32
extends=esp32dev
board=m5stack-core-esp32
upload_speed=115200
build_flags=
    -DDATAPIN=26
    -DNUMLEDS=37`

This **does** compile, creating  `bootloader.bin`  , `partitions.bin` and `firmware.bin` files.

Next, I flashed the `firmware.bin` file, to ` 0x0000`.  Tried to boot, nothing happens.

Next, I flashed `bootloader.bin`  to ` 0x1000`, `partitions.bin` to `0x8000` and `firmware.bin` to `0x10000`.  Again, tried to boot, nothing happens.

I know that the board works as I was previously, successfully, using it as a Bluetooth proxy for my HomeAssistant setup.  As I no longer require the Bluetooth proxy, it was my hope to repurpose the board for this project.

Would you please advise?

Thank you,

Burkert

toblum, 2025-08-15:
Hi Alex,

I'm glad that you're continuing to use the project and want to transfer it to another platform.
At first glance, what you've tried looks good. Personally, I've always uploaded directly to VSCode instead of uploading the files separately. Otherwise, the configuration looks good to me.
I think the main problem is that the ESP32-C3 is a single-core chip, but the code is written to use both cores of the ESP32. So it probably won't work without some adjustments.

Best regards,
Tobias

alexburkert, 2025-08-16:
@toblum 

Thank you for your response.  I will attempt this project on a different board.

Thank you,

Burkert

alexburkert, 2025-08-19:
@toblum 

I uploaded your firmware.bin to an ESP32-S3-WROOM-1. While it does light up, it does not present a wifi hotspot.

I then tried building via platform.io, which completed.  Again, while it does light up, it does not present a wifi hotspot.

Following is the platform.io file that I compiled : 

```
;PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
src_dir = ./src/
default_envs = esp32-s3-devkitc-1
description = The Microsoft Teams Neopixel Presence Device for ESP32

[env]
platform=espressif32
board=esp32dev
framework=arduino
; upload_port=COM4
monitor_speed=115200
upload_speed=921600
build_flags=
    -DDATAPIN=13
    -DNUMLEDS=35
    ; -DCORE_DEBUG_LEVEL=5
lib_deps=
  IotWebConf@2.3.3
  ArduinoJson@6.21.0
  WS2812FX@1.4.1

[env:esp32doit-devkit-v1]
board=esp32doit-devkit-v1

[env:esp32-s3-devkitc-1]
board = esp32-s3-devkitc-1
build_flags=
    ${env.build_flags}

[env:esp32doit-devkit-v1-nocertcheck]
board=esp32doit-devkit-v1
build_flags=
    ${env.build_flags}
    -DDISABLECERTCHECK

[env:m5stack-core-esp32]
platform=espressif32
extends=esp32dev
board=m5stack-core-esp32
upload_speed=115200
build_flags=
    -DDATAPIN=26
    -DNUMLEDS=37
```

I welcome **any** advice which you may have.

Thank you,

Burkert

toblum, 2025-08-19:
Hi!

The error `rmt(legacy): CONFLICT! driver_ng is not allowed to be used with the legacy driver` seems to be the cause. The WS2812FX library uses probably the legacy RMT implementation and that conflicts with the newer RMT implementation from the S3 core.
Not sure if there are more recent verions of WS2812FX you might try to use or different version of the ESP core `platform = espressif32@<someversion>`. But theres not much more I can say.

Greetings

alexburkert, 2025-08-20:
@toblum 

I've tried the following combination and now am presented with the wifi hotspot.

```
[env:esp32-s3-devkitc-1]
platform = espressif32@6.10.0
framework=arduino
lib_deps=
  IotWebConf@2.3.3
  ArduinoJson@6.21.0
  WS2812FX@1.4.5
board = esp32-s3-devkitc-1
monitor_speed=115200
upload_speed=921600
build_flags=
    -DDATAPIN=13
    -DNUMLEDS=35
```

I'm not 100% certain that `platform = espressif32@6.10.0` is required.  However, it does seem that `lib_deps=WS2812FX@1.4.5` _may_ be required.

Hope that this can help another user with the same board.

Thank you,

Burkert

closed 2025-08-20 by alexburkert